WildernessLabs/Meadow.CLI #40: SerialPort problem in AttemptToReconnectToMeadow() with old win7 usbser.sys driver

Dblm0, 2020-12-17:
Using `Windows 7 x64` with `usbser.sys 6.1.7601.17514(win7sp1_rtm.101119-1850)` driver I faced a problem with device reconnection after its reset.

The  `System.IO.FileNotFoundException` is thrown on attempt to close the serial port.

CLI output on --MonoEnable cmd
```
Opening port 'COM32'
Data: Mono enabled, will run app.exe
App: Initialize hardware...
ID: Meadow by Wilderness Labs, Model: F7Micro, MeadowOS Version: 0.4.2 (Dec  8 2020 10:09:54),
Processor: STM32F777IIK6, Processor Id: 37-00-3d-00-0d-51-38-32-37-35-36-30,
Serial Number: 307335783238, CoProcessor: ESP32, CoProcessor OS Version: 0.4.2.0
App: Cycle colors...
Data: Mono has been enabled. Restarting F7 Micro
Host Serial Reconnect
Exception: System.IO.FileNotFoundException: Unable to find the specified file.
   at System.IO.Ports.SerialStream.Dispose(Boolean disposing)
   at System.IO.Stream.Close()
   at System.IO.Ports.SerialPort.Dispose(Boolean disposing)
   at System.ComponentModel.Component.Dispose()
   at System.IO.Ports.SerialPort.Close()
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.Initialize(Boolean listen) i
n C:\Users\Dmitriev\Desktop\C#Projects\Meadow.CLI\Meadow.CLI.Core\DeviceManageme
nt\MeadowSerialDevice.cs:line 87
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.AttemptToReconnectToMeadow()
 in C:\Users\Dmitriev\Desktop\C#Projects\Meadow.CLI\Meadow.CLI.Core\DeviceManage
ment\MeadowSerialDevice.cs:line 429
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.DataReceived(Object sender,
MeadowMessageEventArgs args) in C:\Users\Dmitriev\Desktop\C#Projects\Meadow.CLI\
Meadow.CLI.Core\DeviceManagement\MeadowSerialDevice.cs:line 365
   at MeadowCLI.Hcom.MeadowSerialDataProcessor.ParseAndProcessReceivedPacket(Byt
e[] receivedMsg, Int32 receivedMsgLen) in C:\Users\Dmitriev\Desktop\C#Projects\M
eadow.CLI\Meadow.CLI.Core\Internals\MeadowComms\MeadowSerialDataProcessor.cs:lin
e 323
Exception communicating with Meadow: 
Exception of type 'MeadowCLI.DeviceManagement.MeadowDeviceManagerException' was thrown.
```

Dblm0, 2020-12-28:
I've discovered the same kind of problem with `raspberry PI 3B+` under `Raspbian buster`

```
pi@raspberrypi:~/Meadow.CLI $ ./Meadow.CLI --MonoEnable -s /dev/ttyACM0
Opening port '/dev/ttyACM0'
App: Initialize hardware...
Data: Mono enabled, will run app.exe
App: Cycle colors...
Data: Mono has been enabled. Restarting F7 Micro
Exception: System.UnauthorizedAccessException: Access to the port '/dev/ttyACM0' is denied.
 ---> System.IO.IOException: No such file or directory
   --- End of inner exception stack trace ---
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.OpenSerialPort(String portName) in ...\Meadow.CLI.Core\DeviceManagement\MeadowSerialDevice.cs:line 318
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.Initialize(Boolean listen) in ...\Meadow.CLI.Core\DeviceManagement\MeadowSerialDevice.cs:line 91
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.AttemptToReconnectToMeadow() in ...\Meadow.CLI.Core\DeviceManagement\MeadowSerialDevice.cs:line 430
   at MeadowCLI.DeviceManagement.MeadowSerialDevice.DataReceived(Object sender, MeadowMessageEventArgs args) in ...\Meadow.CLI.Core\DeviceManagement\MeadowSerialDevice.cs:line 366
   at MeadowCLI.Hcom.MeadowSerialDataProcessor.ParseAndProcessReceivedPacket(Byte[] receivedMsg, Int32 receivedMsgLen) in ...\Meadow.CLI.Core\Internals\MeadowComms\MeadowSerialDataProcessor.cs:line 323
Unexpected error occurred: Exception of type 'MeadowCLI.DeviceManagement.MeadowDeviceManagerException' was thrown.

```

logicethos, 2020-12-28:
@Dblm0 I think your issue is access rights.  On the Raspberry Pi, try:

`sudo usermod -a -G dialout pi`

You will probably need to logout/reboot.

Dblm0, 2020-12-29:
> 
> 
> @Dblm0 I think your issue is access rights. On the Raspberry Pi, try:
> 
> `sudo usermod -a -G dialout pi`
> 
> You will probably need to logout/reboot.

Thanks for the advice. Changing access rights didn't help to solve the problem.
BTW, I don't think it's about privileges, but internal System.IO.Ports issue. It looks like the error occurs when we are trying to handle non-existent port by the time device is restarted.
Closing the port earlier in AttemptToReconnectToMeadow() routine helps well in all provided cases. It can be found in PR #41

logicethos, 2020-12-29:
I wrote some Udev rules for Linux and made a pull request for it to be included in the Meadow documentation.  It was pulled as I recall, but cant find it in the main docs.  I'll be back shortly when i'm in my office.

logicethos, 2020-12-29:
It's here.  It's not listed in the document index.
http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Troubleshooting/Linux/

logicethos, 2020-12-29:
@Dblm0 I think that will fix your problem.  If not, please open a new ticket, as I believe this issue is related to Linux.

Dblm0, 2020-12-30:
@logicethos I'm sorry if I introduced a bit of ambiguity by renaming this issue. 
The story is pretty simple. It started from being sure, it's `windows 7` related issue. After finding out the same kind of problem with `linux-arm` and `linux-x64` platforms, I decided to rename this ticket as I think, the issue is common. Only two things made me think that way. 
- First, it happens in AttemptToReconectToMeadow() function. Even thought it throws different exceptions on `windows` and `linux` machines, I believe it's the same thing.
- Second, the provided PR solve them all

Speaking of `linux` and access rights, which may cause device communication issues, I'm afraid it's not exactly my case. There are no particular problems with other CLI command like `--ListFiles` or `--GetDeviceName`, because they don't include reconnect procedure. `--MonoEnable` command, which I use in my tests, does that.

logicethos, 2020-12-30:
@Dblm0 did you try the Udev rules in my previous post?

Dblm0, 2020-12-31:
> 
> 
> @Dblm0 did you try the Udev rules in my previous post?

Yes, I did, but nothing has changed.

thegreatco, 2021-07-12:
This should be fixed in `0.14.0` and newer. Please create a new issue if it isn't.

closed 2021-07-12 by thegreatco