WildernessLabs/Meadow.CLI #43: Timeout communicating with Meadow

Perun85, 2020-12-23:
I am experiencing issues while trying to update ESP32 coprocessor using CLI.

Version: Meadow.CLI 0.12.3

According to console, step where Mono runtime is disabled went well.

[image]

Same goes for runtime update.

[image]

But, when I try to update coprocessor I get following error.

[image]

Perhaps I am doing something wrong?

thegreatco, 2021-07-13:
This should be fixed in the CLI rewrite, please pull the latest and let me know if you still encounter this issue

closed 2021-07-13 by thegreatco

Perun85, 2021-07-18:
With the latest CLI version it is working without issues.

thegreatco, 2021-07-18:
@Perun85 thanks for the update!