WildernessLabs/Meadow.CLI #89: For Investigation: Flash OS issue observed during b5.3 testing

NevynUK, 2021-08-31:
During b5.3 testing there appeared to be some inconsistency with the file loaded on to the board.

`meadow flash os -o Meadow.OS.bin -r Meadow.OS.Runtime.bin`

seemed to write the RT from the directory the command was being executed from but the OS from the download cache.

The final board had 0.5.2.0 of the OS deployed with 0.5.3.0 of the RT.  This is consistent with the file in the cache being 0.5.2 and the files in the local directory being 0.5.3.

NevynUK, 2022-02-06:
This is still a problem with CLI 0.15.1 (beta 0.6.0.1)

NevynUK, 2022-03-08:
I also think this impacts `meadow flash os` as it exhibits the same behaviour.

adrianstevens, 2022-03-08:
@CartBlanche for visibility - let's see if we can improve for B6.3

adrianstevens, 2022-04-10:
@NevynUK I haven't been able to repo on Windows or Mac - lets talk and see if can get a consistent
 repro

NevynUK, 2022-05-31:
Closed following beta 6.4 release.

closed 2022-05-31 by NevynUK