WildernessLabs/Meadow.CLI #93: Meadow auto-connect can still be unreliable

alexischr, 2021-09-23:
Despite huge autoconect improvements in the CLI, I still typically run meadow with the parameter -s /dev/ttyACM* on Linux to make it reliable. Here is a typical scenario for me where Meadow is found and then "lost":
```
$ meadow mono disable
Connecting to Meadow on /dev/ttyACM1
Meadow StdInfo: Mono is enabled
Meadow StdInfo: Mono has been disabled. Restarting Meadow
Connecting to Meadow on /dev/ttyACM1
Connecting to Meadow on /dev/ttyACM2
Meadow StdInfo: Mono is disabled.
Meadow StdInfo: Mono is disabled
Mono Disabled Successfully
Done!
$ meadow mono enable
Connecting to Meadow on /dev/ttyACM1
Unable to find Meadow.
```

Notice that the `mono disable` invocation searched and found the new path for the Meadow (`/dev/ttyACM2`) when ACM1 didn't work, but `mono enable` not only went back to looking for it on ACM1, it did not fall back to searching after that was not found.

In general I am afraid that using any previously saved Meadow path (without falling back to full search) will be unreliable (what if I closed my laptop between meadow invocations; even button resets change the number in Linux, etc.)

adrianstevens, 2021-11-13:
Ah interesting .... the CLI only caches the command line parameter, it doesn't update the cached serial port if it searches for and finds a different port

adrianstevens, 2022-12-06:
I believe this is fixed - closing issue

closed 2022-12-06 by adrianstevens