WildernessLabs/Meadow.CLI #108: Add support to abort a file transfer.

NevynUK, 2021-11-03:
### Request

Add the ability for the CLI to abort a file transfer if there is a problem writing to the file system.

### Background

There are two different behaviours if there is a problem writing a file to the flash file system:

1 - 0.5.3 - HCOM locks waiting for the file to be written to storage.  This is a problem in stm32_qspi.c code.
2 - Post 0.5.3 - HCOM has been modified to send an error message back to the CLI to indicate there is a problem.

We cannot do anything about 1 as this code is out in the wild.  But we can do something about 2.

HCOM has been modified to send two error messages to CLI:

* Received message HCOM_HOST_REQUEST_TEXT_ERROR, Content: Data packet <packet number> file write failed
* Received message HCOM_HOST_REQUEST_TEXT_ERROR, Content: Abort file transfer

The two messages will be sent for every packet so it will get chatty if we do not abort the transfer.

Not sure what ending the file transfer early will do :)

Latest OS build is required for this, build information will be added when available.

NevynUK, 2021-11-03:
Binaries from build:

#20211103.1 Added QSPI error detection on file write.
on Meadow OS

required to test this feature.

adrianstevens, 2021-11-13:
Thanks @NevynUK !

adrianstevens, 2024-03-22:
fixed in CLI v2.x - closing issue

closed 2024-03-22 by adrianstevens