WildernessLabs/Meadow.CLI #110: meadow app deploy -f App.exe fails

pekkah, 2021-11-08:
```
System.ArgumentException: An item with the same key has already been added. Key: C:\projects\pekkah\meadow-libs\src\Verify.Wifi\bin\Debug\netstandard2.1\App.exe
  at System.Collections.Generic.Dictionary`2.TryInsert(TKey key, TValue value, InsertionBehavior behavior)
  at System.Collections.Generic.Dictionary`2.Add(TKey key, TValue value)
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.<>c__DisplayClass74_0.<<DeployAppAsync>g__AddFile|2>d.MoveNext() in D:\a\1\s\Meadow.CLI.Core\Devices\MeadowLocalDeviceFileManager.cs:577
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.DeployAppAsync(String applicationFilePath, String osVersion, Boolean includePdbs, CancellationToken cancellationToken) in D:\a\1\s\Meadow.CLI.Core\Devices\MeadowLocalDeviceFileManager.cs:598
  at Meadow.CLI.Core.Devices.MeadowDeviceHelper.DeployAppAsync(String fileName, Boolean includePdbs, CancellationToken cancellationToken) in D:\a\1\s\Meadow.CLI.Core\Devices\MeadowDeviceHelper.cs:270
  at Meadow.CLI.Commands.App.DeployAppCommand.ExecuteAsync(IConsole console) in D:\a\1\s\Meadow.CLI\Commands\App\DeployAppCommand.cs:40
  at CliFx.CliApplication.RunAsync(ApplicationSchema applicationSchema, CommandInput commandInput) in \home\runner\work\CliFx\CliFx\CliFx\CliApplication.cs:147
  at CliFx.CliApplication.RunAsync(IReadOnlyList`1 commandLineArguments, IReadOnlyDictionary`2 environmentVariables) in \home\runner\work\CliFx\CliFx\CliFx\CliApplication.cs:191
```
It seems that the file App.exe is on both `binaries` list and `dependencies` list at lines 590 and 596 of `MeadowLocalDeviceFileManager`. Then app tries to add it twice to the `files` dictionary where the error above gets thrown.

adrianstevens, 2021-11-13:
Thanks @pekkah  - I wasn't able to reproduce with the latest code so this _may_ be fixed in the next version.

Does it work if you specify app.dll instead of app.exe?

pekkah, 2021-11-14:
Yes, it works with app.dll

adrianstevens, 2022-04-11:
great - closing issue

closed 2022-04-11 by adrianstevens