WildernessLabs/Meadow.CLI #126: App Deploy generates an error if OS files cannot be located

NevynUK, 2022-01-28:
When OS version is set to an internal version then you can get the following error for a `meadow app deploy` request:

```
meadow app deploy -i -f App.dll                                                                                                                                                                                                                                                               ✔  09:12:07
Connecting to Meadow on /dev/tty.usbmodem3067357532381
Download version file for release 0.6.0.8
ERROR
System.Net.Http.HttpRequestException: Response status code does not indicate success: 404 (Not Found).
```

So on the board I have version 0.6.0.8 and I am trying to perform an application deployment.  Note that the flash is empty.

adrianstevens, 2022-01-28:
@NevynUK  this is expected behavior - the version file for 0.6.0.8 obviously isn't hosted / updated for internal builds

NevynUK, 2022-01-28:
Cough - Exception or a user friendly error message ?

adrianstevens, 2022-01-29:
should be a user-friendly message :) I tested the extension - let me try the ol CLI

CartBlanche, 2022-01-30:
@adrianstevens Please link the PR to this issue, once it's ready for review, just so we can keep track of things.

adrianstevens, 2022-04-11:
Fixed here: https://github.com/WildernessLabs/Meadow.CLI/pull/145 

closing issue

closed 2022-04-11 by adrianstevens