WildernessLabs/Meadow.CLI #142: Meadow version check thinks 0.6.2.0 is less than 0.6.1.0

NevynUK, 2022-03-19:
Just received the following message when upgrading a board:

```
Meadow StdInfo: Mono will not start. Mono version is not available (Meadow.OS version 0.6.2.0).
Your OS version is older than 0.6.1.0. A bulk flash erase is highly recommended.
Proceed? (Y/N) Press Y to erase flash, N to continue install without erasing
```

Upgrading to 0.6.3.0 using CLI 0.19.1.

adrianstevens, 2022-04-11:
Fixed with this commit: 
https://github.com/WildernessLabs/Meadow.CLI/commit/ae12add13c32415179c04f157f2218510092d627

Specifically changed the wording to:
Your OS or runtime version is older than {MINIMUM_OS_VERSION}. A flash erase is highly recommended.

NevynUK, 2022-05-31:
Closed following beta 6.4 release.

closed 2022-05-31 by NevynUK