WildernessLabs/Meadow.CLI #150: Unable to connect to meadow com port (version v0.19.1)

brucecrombie, 2022-04-23:
Posted to the Slack support channel on April 3rd, 2022 with no replies to date. Reposting as an issue now.

`meadow use port [comXX] `and `meadow [ command] -s [comxx]` always fails with error message:
`"Selected serial port [comXX] does not exist. Use 'meadow list ports' to view available options and 'meadow use port' to select a valid port"`

`meadow list ports` correctly lists the com ports

The following remedies have been tried:
- deleting all the com ports  to let them regenerate when I connect the meadow
- uninstall the meadow CLI and reinstall
- use Zadig to ensure the correct com port driver

This problem is the same on all 3 meadow F7v1s that I have

dfu-util correctly lists the device
Found DFU: [0483:df11] ver=2200, devnum=18, cfg=1, intf=0, path="1-2.3", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="346438663536"
Found DFU: [0483:df11] ver=2200, devnum=18, cfg=1, intf=0, path="1-2.3", alt=2, name="@OTP Memory /0x1FF0F000/01*001Ke,01*032 e", serial="346438663536"
Found DFU: [0483:df11] ver=2200, devnum=18, cfg=1, intf=0, path="1-2.3", alt=1, name="@Option Bytes  /0x1FFF0000/01*032 e", serial="346438663536"
Found DFU: [0483:df11] ver=2200, devnum=18, cfg=1, intf=0, path="1-2.3", alt=0, name="@Internal Flash  /0x08000000/04*032Kg,01*128Kg,07*256Kg", serial="346438663536"

meadow flash os command gets to the point where it is to connect to the com port and then fails with an unhandled exception - System.TimeoutException: Timeout while writing to serial port
everything worked fine up till I CLI version v0.19.1

adrianstevens, 2022-04-23:
@CartBlanche for visibility

CliffAgius, 2022-05-03:
I have the same issue here.  DFU can see the devices when I use `dfu-util --list` and I can flash the devices Ok but after it flashes the devices are not COM ports but appear as USB devices so the connection fails.

```
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0000
Match product ID from file: 0000
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 2097152

Erase    done.

Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
Warning: Invalid DFU suffix signature
Device Flashed.
Connecting to Meadow on COM4
Connecting to Meadow on COM3
Unhandled exception. System.AggregateException: One or more errors occurred. (Timeout while writing to serial port)
 ---> System.TimeoutException: Timeout while writing to serial port
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.<>c.<EncodeAndSendPacket>b__57_0() in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 277
   at System.Threading.CancellationToken.<>c.<.cctor>b__26_0(Object obj)
   at System.Threading.CancellationTokenSource.CallbackNode.<>c.<ExecuteCallback>b__9_0(Object s)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.CancellationTokenSource.CallbackNode.ExecuteCallback()
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   --- End of inner exception stack trace ---
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.NotifyCancellation(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.<>c.<.cctor>b__56_0(Object obj)
   at System.Threading.TimerQueueTimer.CallCallback(Boolean isThreadPool)
   at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
   at System.Threading.TimerQueue.FireNextTimers()
```

In the Windows Device Manager the MEadow Boards all V1 appear as USB devices 

[image]

the 2 listed COM ports remain after unplugging the Meadow Boards so proven not to be the Meadow board but the USB device does disappear.

CartBlanche, 2022-05-06:
@CliffAgius, @brucecrombie I recently pushed an enhancement for our Windows Extension `develop` branch that filters on just Wilderness Labs devices, in the IDE. Would you like to beta test that version to see if it only allows you to pick up Meadow devices for debugging?

CliffAgius, 2022-05-06:
I would but I am set-up for a conference talk on Monday at NDC London so while I have it working at present I don't want to mess just in case but happy to test after the talk on Monday.

For info the way that I got my set-up working again was to follow the blog here:

https://blog.wildernesslabs.co/b6-2-is-live-with-new-meadow-os-improvements-and-m-f-drivers/

The steps I was missing was to Disable Mono and Flash Erase before flashing the OS and now when I follow those steps it works perfect every time... Well so far anyway...

CartBlanche, 2022-05-06:
@CliffAgius Perfectly understandable. I'm actually based in London. Will this talk be streamed and can the team Tweet about it??

CliffAgius, 2022-05-06:
Won't be streamed but will be available on YouTube about 3 months after the event and yes you can tweet about it details here 

https://ndclondon.com/agenda/net-on-tiny-iot-meadow-boards-0tu9/0t8pnr7rfp7

CartBlanche, 2022-10-11:
@CliffAgius thanks for reminding us.

closed 2022-10-11 by CartBlanche