WildernessLabs/Meadow.CLI #163: Exception when checking for updates for none published firmware

NevynUK, 2022-05-31:
So I have a local copy of the firmware deployed with version number 0.6.3.8 and `meadow mono disable` generates this exception:

```
Error checking for updates
System.Net.Http.HttpRequestException: Response status code does not indicate success: 404 (Not Found).
   at System.Net.Http.HttpResponseMessage.EnsureSuccessStatusCode()
   at System.Net.Http.HttpClient.GetStringAsyncCore(HttpRequestMessage request, CancellationToken cancellationToken)
   at Meadow.CLI.Core.DownloadManager.CheckForUpdatesAsync() in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\DownloadManager.cs:line 251
Connecting to Meadow on /dev/tty.usbmodem3354336F30361
```

Internal OS version so unlikely to impact customers.

adrianstevens, 2022-06-06:
Ah interesting ... this the code that checks for new versions of the CLI, it has nothing to do with the OS version on the board. I'll make fail gracefully :)

patridge, 2022-07-03:
I hit the same issue on `meadow download os`. 

In fairness to the error, if I make the same request in that method via web browser, I also get a 404: https://api.nuget.org/v3-flatcontainer/WildernessLabs.Meadow.CLI/index.json [[referenced here](https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI.Core/DownloadManager.cs#L240)].

Even weirder, the second run ran fine, somehow.

> % meadow download os
> Downloading latest version file
> Downloaded 285 bytes
> OS version 0.6.4.0 is already installed.
> Done!

patridge, 2022-07-03:
I found someone else with a link to that system. Based on their URL, it looks like the URLs might require all-lowercase URLs. Might just need to tweak the package ID to work, since this URL returns a version list.
https://api.nuget.org/v3-flatcontainer/wildernesslabs.meadow.cli/index.json

patridge, 2022-07-03:
Here is a PR with the URL fix in #170, but it should be tested with the correct URL to make sure something else after the check doesn't blow up for users.

closed 2022-12-06 by adrianstevens