WildernessLabs/Meadow.CLI #171: Deployment of files locks on CCM with ethernet enabled

NevynUK, 2022-07-05:
Getting a problem with the CCM when trying to write files to the board.  I get the following exception when writing to the board:

```
Connecting to Meadow on /dev/tty.usbmodem3360317434381
Writing App.dll as App.exe to partition 0
Starting File Transfer...
Meadow StdInfo: File 13% downloaded
Meadow StdInfo: File 21% downloaded
Meadow StdInfo: File 30% downloaded
Meadow StdInfo: File 43% downloaded
Meadow StdInfo: File 52% downloaded
Meadow StdInfo: File 60% downloaded
Meadow StdInfo: File 73% downloaded
Meadow StdInfo: File 82% downloaded
Meadow StdInfo: File 91% downloaded
Exception sending command to Meadow
RequestType: HCOM_MDOW_REQUEST_END_FILE_TRANSFER Timeout: 00:01:00 UserData: 1 ResponseType none CompletionMessageType: none ResponseHandler: False CompletionPredicate: True ResponsePredicate: True
Meadow.CLI.Core.Exceptions.MeadowCommandException: Command timeout waiting for response.
 ---> System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 372
   --- End of inner exception stack trace ---
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 378
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendCommandAsync(Command command, CancellationToken cancellationToken, String caller) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 209
   at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendTheEntireFile(FileCommand command, Boolean lastInSeries, CancellationToken cancellationToken) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:line 134
ERROR
Meadow.CLI.Core.Exceptions.MeadowCommandException: Command timeout waiting for response.
    System.Threading.Tasks.TaskCanceledException: A task was canceled.
      at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:372
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.WaitForResponseMessageAsync(Command command, CancellationToken cancellationToken, String caller) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:378
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendCommandAsync(Command command, CancellationToken cancellationToken, String caller) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:209
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendTheEntireFile(FileCommand command, Boolean lastInSeries, CancellationToken cancellationToken) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceComms.cs:134
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.WriteFileAsync(String sourceFileName, String destinationFileName, TimeSpan timeout, CancellationToken cancellationToken) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDeviceFileManager.cs:134
  at Meadow.CLI.Commands.Files.WritesFileCommand.ExecuteAsync(IConsole console) in /D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI\Commands\Files\WriteFileCommand.cs:78
  at CliFx.CliApplication.RunAsync(ApplicationSchema applicationSchema, CommandInput commandInput) in /D:\a\CliFx\CliFx\CliFx\CliApplication.cs:147
  at CliFx.CliApplication.RunAsync(IReadOnlyList`1 commandLineArguments, IReadOnlyDictionary`2 environmentVariables) in /D:\a\CliFx\CliFx\CliFx\CliApplication.cs:191
Done!
Connecting to Meadow on /dev/tty.usbmodem3360317434381
Unable to find Meadow.
```

I have a `meadow.config.yaml` deployed to the board with the following content:

```yaml
Network:
    Ethernet:
        Default: true
        IPAddress: 192.168.1.60
        NetMask: 255.255.255.0
        Gateway: 192.168.1.254
    WiFi:
        Default: false
    GetNetworkTimeAtStartup: 1
    NtpServers:
        - 0.uk.pool.ntp.org
        - 1.uk.pool.ntp.org
        - 2.uk.pool.ntp.org
        - 3.uk.pool.ntp.org
    NtpRefreshPeriod: 3600
    DnsServers:
        - 1.1.1.1
Debug:
    Uart1Use: trace
    TraceLevel: 1

```

Note that the file looks to be written OK as I am seeing this when I turn on UART output from the STM32:

```
[    0.287000] Ethernet is being initialized
[    0.287000] (Note) Meadow 0.6.5.13 (Jul  5 2022@14:45:11) rebooted, H/W:CCMv2, Mono:Disabled, Trace level:0x7f, to:UART1 (ramlog)
[    0.291000] (Warn) mono/hcom_mono_control.c@427-Mono is disabled.
[    1.256000] ESP32 Coprocessor ready, firmware version 0.6.5.2
[    2.361000] Ethernet active MAC:4a:5a:6a:09:98:08, IP:192.168.1.60
[    5.363000] Link Status of PHY 1 is Up
[    5.363000] Getting time from 0.uk.pool.ntp.org
[    5.384000] INFO: '0.uk.pool.ntp.org' resolved to: 5.101.146.245
[    5.409000] ToDo: Ethernet updated time, Mono needs to be notified
[   39.110000] (Info) file/hcom_file_dnld_proc.c@183-Meadow downloading file (FileLen:11776, Crc:0xdfb36537, Name:App.exe)
[   39.278000] (Note) End of file transfer
```

Steps:
* Deploy a config file as above to the board (may need to adjust IP address etc for your configuration)
* Connect to a wired network
* Reset the board and wait a short time
* From a directory with compiled binaries execute the command `meadow file write -f App.dll -t App.exe`

The system should write the file and then lock, eventually timing out.  The file being written does not have to be the `App.dll`, it can be the config file itself.

Resetting the board allows you to write files and disable the application etc.

NevynUK, 2022-07-06:
Forgot to add to the original report, the serial port is still listed in the available ports on the Mac.

adrianstevens, 2024-03-22:
Fixed in CLI v2.x - closing issue

closed 2024-03-22 by adrianstevens