WildernessLabs/Meadow.CLI #223: Request: Extend list ports to use a port

NevynUK, 2022-11-25:
Add an option to `meadow list ports` to use a port if there is only one meadow device connected.

Adding an additional option to use the port would ease the case where the user issues the commands

* `meadow list ports`
* `meadow use port xxxxxxx`

adrianstevens, 2022-11-26:
Maybe we configure 'meadow use port' to work without an argument if there's only one port?

I feel like selecting a port when calling list is overfunctioning - pun intended ;)

CartBlanche, 2022-11-28:
@NevynUK there was some talk about streamlining this experience. 
eg. If there is only 1 device present, the user should not have to select it, it should detect there is only 1 and do some magic under the hood to auto-select and use it. 
If there is more than 1 device attached, then it can pull up a dynamically numbered list with each device and the user can select the device via the command line prompt and it automagically uses that device and continues.
Those were the original thoughts anyway. Just less paper cuts in the CLI experience.

adrianstevens, 2022-12-06:
@CartBlanche we should sync and talk strategy

adrianstevens, 2024-03-03:
`meadow select port` will automatically select the port if there's only one available in CLI v2

closing issue

closed 2024-03-03 by adrianstevens