WildernessLabs/Meadow.CLI #224: Separation of Device Information Formatting Concerns...

CartBlanche, 2022-11-28:
As per Slack...
Before v1.0 we'd like to separate out the device formatting from the CLI.Core
It currently resides - https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI.Core/DeviceManagement/MeadowDeviceInfo.cs#L357

@adrianstevens Suggested we should move the formatting out of CLI.Core and allow/force each UI to format the device information as they see fit.

So we would need to be support each of the following....
- [ ] CLI
- [ ] VS Mac
- [ ] VS Windows
- [ ] VS Code

adrianstevens, 2022-11-28:
And we should include Workbench https://github.com/WildernessLabs/Meadow.Workbench

adrianstevens, 2022-11-28:
Obvious pro is - we should be able to make changes to CLI.Core without breaking the experience on the client side.

Downside of course is adding implementation to each client.

I'm also in agreement with @ctacke - there's no real harm in the clients using the `ToString` method in the short term.

closed 2024-03-22 by adrianstevens