WildernessLabs/Meadow.CLI #255: Duplicate progress data when flashing OS binaries

adrianstevens, 2023-03-09:
CLI 0.96.0-alpha - flash any OS to Meadow to repro

adrianstevens, 2023-03-23:
looks like this:

Meadow StdInfo: Flash 10% complete
Flash 10% complete
Meadow StdInfo: Flash 20% complete
Flash 20% complete
Meadow StdInfo: Flash 30% complete
Flash 30% complete
Meadow StdInfo: Flash 40% complete
Flash 40% complete
Meadow StdInfo: Flash 50% complete
Flash 50% complete
Meadow StdInfo: Flash 60% complete
Flash 60% complete
Meadow StdInfo: Flash 70% complete
Flash 70% complete
Meadow StdInfo: Flash 80% complete
Flash 80% complete

adrianstevens, 2023-03-24:
Just tested the latest code in `develop` and I'm seeing this:

```Meadow StdInfo: Mono will not start - version mismatch: Meadow.OS version 0.9.6.2, Mono version 0.9.6.1
Meadow StdInfo: Mono is disabled
Mono is disabled
Writing C:\Users\Adrian\AppData\Local\WildernessLabs\Firmware\0.9.6.2\Meadow.OS.Runtime.bin runtime
Starting File Transfer...

[
=
=
Meadow StdInfo: File 10% downloaded
=
=
Meadow StdInfo: File 20% downloaded
=
=
Meadow StdInfo: File 30% downloaded
=
=
Meadow StdInfo: File 40% downloaded
=
=
Meadow StdInfo: File 50% downloaded
=
=
Meadow StdInfo: File 60% downloaded
=
=
Meadow StdInfo: File 70% downloaded
=
=
Meadow StdInfo: Meadow successfully deleted '/meadow0/Meadow.OS.Runtime.bin'
An unknown request value of '0x0312' was received.
=
=
=
=
Download of 'Meadow.OS.Runtime.bin' state unknown due to checksum calulation fault:-2
]

Transfer Complete, wrote 3145728 bytes to Meadow

CartBlanche, 2023-03-24:
@adrianstevens this is a UI bug I'm investigating.

adrianstevens, 2024-01-18:
This is fixed in CLI v2

adrianstevens, 2024-03-03:
Shipped - closing issue

closed 2024-03-03 by adrianstevens