WildernessLabs/Meadow.CLI #278: Stdout from Meadow is ASCII instead of UTF-8

bcr, 2023-04-12:
My fancy Meadow app has UTF-8 output. This is turned into ASCII when monitoring with `meadow listen`. I know where to make the change, but I think it may affect other stuff. Works on my machine when using `meadow listen` but will probably explode if presented with non-UTF8 bytes with the high bit set.

https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI.Core/Internals/MeadowCommunication/ReceiveClasses/ReceiveSimpleText.cs#L31

CartBlanche, 2023-04-12:
@bcr what languages are you using?

bcr, 2023-04-12:
English. I was using the Degree Fahrenheit character \u2109.

CartBlanche, 2023-04-12:
@bcr I've create a PR here - https://github.com/WildernessLabs/Meadow.CLI/pull/279 
If you are feeling brave, this will create a Meadow.CLI  artefact here https://github.com/WildernessLabs/Meadow.CLI/actions/runs/4679452989 when finished, which you can install and test as follows, after you've downloaded it an extracted the zip file, grab the nuget from `Meadow.CLI/bin/Release` folder named WildernessLabs.Meadow.CLI.0.97.0-alpha.nupkg...

* dotnet tool uninstall --global WildernessLabs.Meadow.CLI
* dotnet tool install --global --add-source [your-download-dir] WildernessLabs.Meadow.CLI --version 0.97.0-alpha

if you find this alpha version unstable and want to go back to the latest stable version:

* dotnet tool uninstall --global WildernessLabs.Meadow.CLI
* dotnet tool install --global WildernessLabs.Meadow.CLI

bcr, 2023-04-12:
Yeah, that's basically what I did initially and it did what I expected. I submitted #280 which might be a more timid approach.

I just run it with `dotnet run --framework net6.0 -- listen` and it works -- do you prefer that I install it as a tool to test it?

CartBlanche, 2023-04-13:
@bcr if you have the cycles to install it as a tool, that would be great. Just to cover all bases.

CartBlanche, 2023-04-19:
Fixed in #279.

closed 2023-04-19 by CartBlanche