WildernessLabs/Meadow.CLI #287: Decouple linker from deploy

lamebrain, 2023-05-02:
Currently, linking only occurs when deploying to a device. We want to be able to run the linker before creating a package (MPAK).

CartBlanche, 2023-05-02:
Linking should really be part of the build process. 
@lamebrain so do we need to hook into each IDEs "package" feature too?
What about "publish" from IDEs, will we support an app hub/shop in future?

lamebrain, 2023-05-08:
@CartBlanche  Ideally, it's part of the build process, that should be feasible within the IDEs. I normally build through the command line... can we hook into `dotnet build`? If not, we can make a separate CLI command to link out unneeded items.

We'll definitely want to pack and publish from the IDEs in the future. A shop is an interesting idea, we don't have plans for anything like that yet, but definitely Meadow.Cloud for now.

jorgedevs, 2023-08-14:
@CartBlanche any updates on this? this would really improve the experience of using OTA updates without the need to first deploy the app with a machine every time.

CartBlanche, 2023-08-15:
@jorgedevs Once I sort out the VSCode PR, I hope to have a look at this again. Jon already started on it, but due to other priorities, hasn't had a chance to finish that PR. So I'll take a look and see if I can make some progress on it.

adrianstevens, 2024-01-18:
The code has been isolated and cleaned up in CLI v2

adrianstevens, 2024-03-03:
Fixed in v2.0.x - closing issue

closed 2024-03-03 by adrianstevens