WildernessLabs/Meadow.CLI #301: flash doesn't check if flash is already erased, so throws "version is older" error.

CartBlanche, 2023-05-30:
Message from Richard - DoingNZ 
```
The ... Your OS version is older than 0.6.1.0 ... message is due to the flash erase. it's blank and therefore nothing found. Can answer N. Answering Y would erase the flash again. 
[@bryan costanich](https://wildernesslabspublic.slack.com/team/UB9BW8970)
, it would be more intuitive for users if the firmware did a "blank check" during the flash os and skip this prompt if the flash is already erased. The same would apply if a person did multiple flash erase consecutively. Its a trivial bit of code to check all address in the appropriate blocks are erased and executes very quickly in native code.
```

adrianstevens, 2024-01-18:
This is because the flash is erased and it can't determine the OS version.

The check for 0.6.x and older has been removed from CLI v2 - we can close this issue when it goes live.

adrianstevens, 2024-03-15:
Closing issue

closed 2024-03-15 by adrianstevens