WildernessLabs/Meadow.CLI #326: meadow flash os hangs on macOS with CLI 1.1.0

bcr, 2023-07-13:
macOS 12.6.7, Meadow CLI 1.1.0

Subject says it all, I'm in bootloader mode, everything should be fine, but it hangs.

I built my own version from the `develop` branch and that hung also. I added debug logging to trace down the problem. It hangs when calling `device.Open` in `GetDeviceSerial`

```
➜  Meadow.CLI git:(develop) ✗ meadow flash os                          
Flashing OS with /Users/blake/.local/share/WildernessLabs/Firmware/1.1.0.0/Meadow.OS.bin
(hangs)
```

But here's the weird part, I made the following changes and it works fine.

```diff
diff --git a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
index 16582d7..d7bc267 100644
--- a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
+++ b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
@@ -147,7 +147,9 @@ namespace Meadow.CLI.Core.Internals.Dfu
         public static async Task<bool> FlashFile(string fileName, IUsbDevice? device = null, ILogger? logger = null, DfuFlashFormat format = DfuFlashFormat.Percent)
         {
             logger ??= NullLogger.Instance;
+            logger.LogDebug("Device is {device}", device);
             device ??= GetDeviceInBootloaderMode();
+            logger.LogDebug("Device is {device}", device);
 
             if (!File.Exists(fileName))
             {
```

So I don't know if something is getting touched when performing the string-ization of the `device` object for the logging or what. But no problems after that.

I tried putting in a `Thread.Sleep` in case this might be a timing issue (maybe the `device` is so fresh that it is in some weird half-constructed state?) That didn't seem to improve things.

This seems to be environment-specific in some way, and I'm happy to try things out. I have a fork of the `develop` branch on the CLI repo so I think I can reliably build it.

bcr, 2023-07-13:
I found an even easier workaround, but I still don't know what's going on. Has a strong smell of a threading issue or a race condition.

```diff
diff --git a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
index 16582d7..a1abf93 100644
--- a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
+++ b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
@@ -146,6 +146,7 @@ namespace Meadow.CLI.Core.Internals.Dfu
 
         public static async Task<bool> FlashFile(string fileName, IUsbDevice? device = null, ILogger? logger = null, DfuFlashFormat format = DfuFlashFormat.Percent)
         {
+            await Task.Delay(1);
             logger ??= NullLogger.Instance;
             device ??= GetDeviceInBootloaderMode();
 ```

CartBlanche, 2023-07-14:
@bcr are you using an x64 MacOS?

bcr, 2023-07-14:
M1.

```
➜  ~ uname -a
Darwin Blakes-MacBook-Pro.local 21.6.0 Darwin Kernel Version 21.6.0: Thu Jun  8 23:56:13 PDT 2023; root:xnu-8020.240.18.701.6~1/RELEASE_ARM64_T6000 arm64
```

CartBlanche, 2023-07-17:
when you do launch:
`meadow flash os``

How long are you waiting before you say it is hanging?

There can be a delay for 10-20s before the 1st message of:
`10% downloaded`
appears. Keep in mind a Meadow is a 216 MHz (not GHz) device.

We know that isn't a great UX, so we are looking at adding a "working" animation so customers know something is happening.

bcr, 2023-07-24:
It's definitely more than 10 seconds, and I had a log event after the USB device open call that didn't come out.

What I'm thinking right now is that because the initial UsbContext is Dispose'd, that might have an inconsistent effect on the Device that it produces.

I don't have it reproducing right now, but I'll keep an eye on it.

bcr, 2023-07-26:
I got back in the bad state and left it there for a couple of minutes.

```
➜  Meadow.CLI git:(develop) ✗ time dotnet run --framework net6.0 -- flash os
Flashing OS with /Users/blake/.local/share/WildernessLabs/Firmware/1.1.0.0/Meadow.OS.bin
^C^C^CSEHExceptionThread: 1026: CONTEXT_SetThreadContextOnPort (os/kern) aborted
dotnet run --framework net6.0 -- flash os  15.84s user 80.53s system 87% cpu 1:49.83 total
➜  Meadow.CLI git:(develop) ✗ 
```

bcr, 2023-07-26:
This is the direction I'm going in. This works on my machine anyway.

```diff
diff --git a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
index 16582d7..45252d3 100644
--- a/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
+++ b/Meadow.CLI.Core/Internals/Dfu/DfuUtils.cs
@@ -18,6 +18,7 @@ namespace Meadow.CLI.Core.Internals.Dfu
         static int _osAddress = 0x08000000;
         static string _usbStmName = "STM32  BOOTLOADER";
         static int _usbBootLoaderVenderID = 1155; // Equivalent to _usbStmName but for the LibUsbDotNet 3.x
+        static UsbContext usbContext = new UsbContext();
 
         public static string LastSerialNumber { get; private set; } = "";
 
@@ -53,16 +54,13 @@ namespace Meadow.CLI.Core.Internals.Dfu
 
         public static IEnumerable<IUsbDevice> GetDevicesInBootloaderMode()
         {
-            using (UsbContext context = new UsbContext())
+            var allDevices = usbContext.List();
+            var ourDevices = allDevices.Where(d => d.Info.VendorId == _usbBootLoaderVenderID);
+            if (ourDevices.Count() < 1)
             {
-                var allDevices = context.List();
-                var ourDevices = allDevices.Where(d => d.Info.VendorId == _usbBootLoaderVenderID);
-                if (ourDevices.Count() < 1)
-                {
-                    throw new DeviceNotFoundException("No Devices found. Connect a device in bootloader mode. If the device is in bootloader mode, please update the device driver. See instructions at https://wldrn.es/usbdriver");
-                }
-                return ourDevices;
+                throw new DeviceNotFoundException("No Devices found. Connect a device in bootloader mode. If the device is in bootloader mode, please update the device driver. See instructions at https://wldrn.es/usbdriver");
             }
+            return ourDevices;
         }
 
         public static string GetDeviceSerial(IUsbDevice device)
```

This keeps the `UsbContext` around foreva.

In the event that no one else sees this problem, I can understand if you don't want to change it. I think the change is fairly simple, but it's possible that retaining the UsbContext has some side-effect that I haven't seen.

I can submit a PR against current `develop` if you're OK with this.

CartBlanche, 2023-07-27:
@bcr I think having a static `usbContext` may be fine. I think The way to test it would be to create a console app that has a static `usbContext` and as you plug and unplug USB devices the `*.List()` should hopefully show the changes without issue. 

I'm going to test it locally too. The main this is that the change works consistently on macOS, Windows and Linux.

bcr, 2023-07-29:
I'm going to be out of town for a week or so, but I think the important thing is that the context is retained while the device object is being used. So even if GetDevicesInBootloaderMode made a new context each time to stick in the static, that would probably work (any device currently out there would not be active if GetDevicesInBootloaderMode is called again)

I appreciate that you have many platforms to support, so absolutely do the right thing to maximize happiness. In my case, when it gets in the "stuck" state I can't reliably get it to work again, but it sometimes gets there.

bcr, 2023-10-23:
Yes. Just tried with 1.3.4

```
➜  Temp meadow --version                              
v1.3.4
Done!
➜  Temp meadow flash os 
Flashing OS with /Users/blake/.local/share/WildernessLabs/Firmware/1.3.4.0/Meadow.OS.bin
(hangs)
```

Device is in bootloader mode:

```
    |   +-o STM32  BOOTLOADER@03121000  <class IOUSBHostDevice, id 0x10004955f, registered, matched, active, busy 0 (16 ms), retain 22>
```

CartBlanche, 2023-10-23:
@bcr are you still on macOS 12.6.7? Just checking before I pull out my old mac laptop to test 1.3.4 on it.
Could we possibly organise a video all sometime this week to try to get to the bottom of this on your machine?
DM my on our public slack, if you can.

bcr, 2023-10-25:
Using `2.0.0-alpha` is working so far

adrianstevens, 2024-03-03:
Fixed in CLI v2.0.x - closing issue

closed 2024-03-03 by adrianstevens