WildernessLabs/Meadow.CLI #327: -m option doesn't work

bcr, 2023-07-13:
I see in the code that the intent is to allow changing the logging level by specifying the `-m` flag (potentially multiple times to get more and more info.) When it gets into the "real" command line parsing, the option isn't recognized and you will not go to space today.

```
➜  Meadow.CLI git:(develop) ✗ meadow -m device info
Using log level Debug
Unrecognized option(s):
-m
```

I can help fix this -- is the right answer to hammer on the `args` array and remove it, or is the right answer to make the command line parsing recognize it?

CartBlanche, 2023-07-14:
@bcr trace level should be set using `meadow trace level`. Where do you see a -m parameter?

bcr, 2023-07-14:
https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI/Program.cs#L29

bcr, 2023-07-14:
By the way, my intent was to set the logging level for the meadow CLI, not for the Meadow itself.

bcr, 2023-07-14:
It looks like @adrianstevens added this originally with the `-v` flag, then changed it because `-v` was used for a version number for `flash os`

bcr, 2023-10-26:
It looks like this is being revised to `--verbose` -- I'll try it when it comes out.

https://github.com/WildernessLabs/Meadow.CLI/commit/fb256551a4ad8b0ead8df66d012bb2ad28f203c4#diff-b15180c77374b1ef003f717daa86e675a2acf6150221c5c384c4346d1950621dL29

bcr, 2023-11-08:
As of 2.0.0-alpha.1 this is not working

```
➜  ~ meadow --verbose download os
Using log level Verbose
Unrecognized option(s):
--verbose

meadow v2.0.0
```

CartBlanche, 2023-11-08:
@bcr I thought I had fixed that in alpha.1, but I'll make make sure it's ready for alpha.2

bcr, 2023-11-17:
As of 2.0.0-alpha.3 this is not working

```
➜  ~ dotnet tool update WildernessLabs.Meadow.CLI --global --version 2.0.0-alpha.3
Skipping NuGet package signature verification.
Tool 'wildernesslabs.meadow.cli' was successfully updated from version '2.0.0-alpha.1' to version '2.0.0-alpha.3'.
➜  ~ meadow --verbose download os
Using log level Verbose
Unrecognized option(s):
--verbose

meadow v2.0.0
```