WildernessLabs/Meadow.CLI #373: Exit CLI when `shutdown` message is sent stdout

NevynUK, 2023-10-09:
Meadow Core traps application exceptions and generates the message `Meadow StdOut: Shutdown` when an application fails.

It may be desirable if the CLI exited when this message was received as there should be no more output from the application and so nothing to listen for.

CartBlanche, 2023-10-09:
@NevynUK Do you know if Is this sent via HCOM message too?

NevynUK, 2023-10-09:
I don't think there is any message through CLI as the exception is caught in Core.

Would need to review what happens in Core.

adrianstevens, 2024-03-22:
This may not make sense in cases where the Meadow device resets on failure.

Would could potentially make it configurable if this is an important feature for you @NevynUK.