WildernessLabs/Meadow.CLI #398: CLI V2 `meadow esp32 file write` or equivalent does not accept file name

NevynUK, 2023-11-16:
The command `meadow esp32 file write -f ./build/MeadowComms.bin --McuDestAddress 0x10000` on the V1 software would write the file given to the ESP32 at the specified address.  I cannot find an equivalent in the V2 software.

CartBlanche, 2023-11-16:
@NevynUK Is the address likely to change, because we have a GetFileTargetAddress which looks like this...

private int GetFileTargetAddress(string fileName)
        {
            // TODO: determine device type so we can map the file names to target locations
            //       for now we only support the F7 so these are static and well-known

            var fn = Path.GetFileName(fileName).ToLower();
            switch (fn)
            {
                case "meadowcomms.bin":
                    return 0x10000;
                case "bootloader.bin":
                    return 0x1000;
                case "partition-table.bin":
                    return 0x8000;
                default: throw new NotSupportedException($"Unsupported coprocessor file: '{fn}'");
            }
        }

So I suppose the question is, do we definitely want to be able to pass on a specific address in future?

NevynUK, 2023-11-16:
The addresses are unlikely to change in the future.

CartBlanche, 2023-11-16:
@NevynUK So in theory just passing in `-f ./build/MeadowComms.bin` should be enough, as internally it will grab the right address, until we decide to move things around.

NevynUK, 2023-11-16:
Yes, that is correct.  It will be highly unlikely that we will move the files though.

PR #399 (2023-11-16): Allow esp command to accept file path
Fixes #398
closed 2023-11-16 by CartBlanche